ivmarkov/rust-esp32-std-demo #129: Unrecognized command line option '--target=xtensa-esp32-espidf'

3d-t, 2022-11-18:
When I run 'cargo build' on MacBook pro with Big Sur 11.7 I get the following error:

xtensa-esp32-elf-gcc: error: unrecognized command line option '--target=xtensa-esp32-espidf'
ninja: build stopped: subcommand failed.

ivmarkov, 2022-11-18:
As a workaround, unset `CC=clang` from your environment.

3d-t, 2022-11-18:
Thanks for your quick reply but 'unset CC' did not solve the issue:

   Compiling async-net v1.7.0
   Compiling async-fs v1.6.0
   Compiling uncased v0.9.7
error: failed to run custom build command for `esp-idf-sys v0.31.11`

Caused by:

      xtensa-esp32-elf-g++: error: unrecognized command-line option '--target=xtensa-esp32-espidf'
      ninja: build stopped: subcommand failed.

ivmarkov, 2022-11-18:
I'm relatively sure that's the reason. Maybe it is not (only) the CC env variable but some additional ones. Coincidentally, we had another user complaining about this and for him it worked by unsetting CC (and maybe a few others).

Can you go to the forum and search for the relevant messages?

The background of this issue, is that `esp-idf-sys` internally builds ESP IDF itself, and for that, it **must** use the xtensa GCC toolchain. However, if the CC env var (and maybe a few others) are set to "clang", CMake (which drives the ESP IDF build) mistakenly assumes that the xtensa GCC compiler is clang (while it is not!) and passes to it the invalid `--target` option.

ivmarkov, 2022-11-18:
Yeah, a proper fix *might* be that we are ourselves unsetting these vars during the `esp-idf-sys` build somehow, but in the meantime, user unsetting those should help.

ivmarkov, 2022-11-18:
Here's the relevant message from the forum:
`I figured it out 🎉 CXX and CC env variables where confusing cmake, it checked those for default flags and thought it was using clang, not gcc, when I cleared them things worked`

ivmarkov, 2022-11-20:
Ping?

closed 2022-11-20 by ivmarkov

3d-t, 2022-11-20:
Thanks for your replies. After unsetting both CXX and CC it now proceeds but there is another error:

    Tool doesn't match supported version from list ['esp-2022r1-11.2.0']:
    [USER]/.espressif/tools/xtensa-esp32-elf-gcc/8_4_0-esp-2021r2-patch3-x86_64-apple-darwin/bin/xtensa-esp32-elf-gcc

    Please try to run 'idf.py fullclean' to solve it.

After running 'idf.py fullclean'  the same error message.

3d-t, 2022-11-20:
Hi ivmarkov,

Thanks for your replies and sorry for not responding earlier. It is still not working and I added a comment. 

Regards,,

Maarten van Emmerik

3d-t, 2022-11-20:
Ivan,

You can ignore my comment. I deleted the ~/.rustup and ~/.espressif directories and installed everything from scratch:

```
curl -LO https://github.com/esp-rs/rust-build/releases/download/v1.65.0.1/install-rust-toolchain.sh
install-rust-toolchain.sh -s master
. ./export-esp.sh
cargo install ldproxy
rustup default esp
unset CC
unset CXX
git clone https://github.com/ivmarkov/rust-esp32-std-demo
cd rust-esp32-std-demo
cargo build

```

The error described in this issue does not appear anymore but now

```
error[E0308]: mismatched types
  --> /Users/maarten/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.11/src/lib.rs:36:62
   |
36 | const ESP_IDF_TIME64_CHECK: ::std::os::espidf::raw::time_t = 0 as crate::time_t;
   |                                                              ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`

error[E0308]: mismatched types
  --> /Users/maarten/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.11/src/lib.rs:38:51
   |
38 | const ESP_IDF_TIME64_CHECK_LIBC: ::libc::time_t = 0 as crate::time_t;
   |                                                   ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`

For more information about this error, try `rustc --explain E0308`.
```

I created a new issue for this error:

Axxem B.V.
Maarten van Emmerik
Mgr. De Haasstraat 9
5521 TA Eersel
***@***.***