ivmarkov/rust-esp32-std-demo #130: Mismatched types error

3d-t, 2022-11-20:
After running 'cargo build' I get the following error:

Compiling esp-idf-sys v0.31.11
error[E0308]: mismatched types
  --> /Users/maarten/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.11/src/lib.rs:36:62
   |
36 | const ESP_IDF_TIME64_CHECK: ::std::os::espidf::raw::time_t = 0 as crate::time_t;
   |                                                              ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`

error[E0308]: mismatched types
  --> /Users/maarten/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-sys-0.31.11/src/lib.rs:38:51
   |
38 | const ESP_IDF_TIME64_CHECK_LIBC: ::libc::time_t = 0 as crate::time_t;
   |                                                   ^^^^^^^^^^^^^^^^^^ expected `i32`, found `i64`

For more information about this error, try `rustc --explain E0308`.
error: could not compile `esp-idf-sys` due to 2 previous errors

ivmarkov, 2022-11-20:
You are trying to build with ESP IDF 5.0 / master - yet - you have not enabled the [espidf_time64](https://github.com/ivmarkov/rust-esp32-std-demo/blob/main/.cargo/config.toml#L11) flag.

ivmarkov, 2022-11-20:
Here, the gory details: https://github.com/esp-rs/rust/issues/110
:)

closed 2022-11-20 by ivmarkov