ivmarkov/rust-esp32-std-demo #131: struct `esp_idf_sys::eth_dm9051_config_t` has no field named `spi_hdl`

3d-t, 2022-11-21:
Error after running `cargo build`:

``` 
Compiling mipidsi v0.5.0
   Compiling esp-idf-hal v0.39.1
   Compiling esp-idf-svc v0.43.1
error[E0560]: struct `esp_idf_sys::eth_dm9051_config_t` has no field named `spi_hdl`
   --> /Users/maarten/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.43.1/src/eth.rs:357:21
    |
357 |                     spi_hdl: spi_handle as *mut _,
    |                     ^^^^^^^ `esp_idf_sys::eth_dm9051_config_t` does not have this field
    |
    = note: available fields are: `spi_host_id`, `spi_devcfg`, `int_gpio_num`

error[E0560]: struct `esp_idf_sys::eth_w5500_config_t` has no field named `spi_hdl`
   --> /Users/maarten/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.43.1/src/eth.rs:371:21
    |
371 |                     spi_hdl: spi_handle as *mut _,
    |                     ^^^^^^^ `esp_idf_sys::eth_w5500_config_t` does not have this field
    |
    = note: available fields are: `spi_host_id`, `spi_devcfg`, `int_gpio_num`
```

config.toml

```
[build]
# Uncomment the relevant target for your chip here (ESP32, ESP32-S2, ESP32-S3 or ESP32-C3)
target = "xtensa-esp32-espidf"
#target = "xtensa-esp32s2-espidf"
#target = "xtensa-esp32s3-espidf"
#target = "riscv32imc-esp-espidf"

[target.xtensa-esp32-espidf]
linker = "ldproxy"
# Uncomment for ESP-IDF 5 and later
rustflags = ["--cfg", "espidf_time64"]

[target.xtensa-esp32s2-espidf]
linker = "ldproxy"
# Uncomment for ESP-IDF 5 and later
#rustflags = ["--cfg", "espidf_time64"]

[target.xtensa-esp32s3-espidf]
linker = "ldproxy"
# Uncomment for ESP-IDF 5 and later
#rustflags = ["--cfg", "espidf_time64"]

[target.riscv32imc-esp-espidf]
linker = "ldproxy"
# Uncomment for ESP-IDF 5 and later
#rustflags = ["--cfg", "espidf_time64", "-C", "default-linker-libraries"]
# Future - necessary for the experimental "native build" of esp-idf-sys with ESP32C3
# See also https://github.com/ivmarkov/embuild/issues/16
rustflags = ["-C", "default-linker-libraries"]

[unstable]
build-std = ["std", "panic_abort"]
#build-std-features = ["panic_immediate_abort"] # Only necessary if building against ESP-IDF tag `v4.3.2` (the minimum supported version)

[env]
# Uncomment this and use the esp-idf-sys "native" build feature (`cargo build --features native`) to build against ESP-IDF 5.0 (master)
ESP_IDF_VERSION = { value = "master" }

# Comment out this when using the PlatformIO build (it only supports `v4.3.2`)
#ESP_IDF_VERSION = { value = "release/v4.4" }

# These configurations will pick up your custom "sdkconfig.release", "sdkconfig.debug" or "sdkconfig.defaults[.*]" files
# that you might put in the root of the project
# The easiest way to generate a full "sdkconfig" configuration (as opposed to manually enabling only the necessary flags via "sdkconfig.defaults[.*]"
# is by running "cargo pio espidf menuconfig" (that is, if using the pio builder)
#ESP_IDF_SDKCONFIG = { value = "sdkconfig.release;sdkconfig.debug" }
ESP_IDF_SDKCONFIG_DEFAULTS = { value = "sdkconfig.defaults;sdkconfig.defaults.esp32;sdkconfig.defaults.esp32s2" }

--- 
 Compiling mipidsi v0.5.0
   Compiling esp-idf-hal v0.39.1
   Compiling esp-idf-svc v0.43.1
error[E0560]: struct `esp_idf_sys::eth_dm9051_config_t` has no field named `spi_hdl`
   --> /Users/maarten/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.43.1/src/eth.rs:357:21
    |
357 |                     spi_hdl: spi_handle as *mut _,
    |                     ^^^^^^^ `esp_idf_sys::eth_dm9051_config_t` does not have this field
    |
    = note: available fields are: `spi_host_id`, `spi_devcfg`, `int_gpio_num`

error[E0560]: struct `esp_idf_sys::eth_w5500_config_t` has no field named `spi_hdl`
   --> /Users/maarten/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.43.1/src/eth.rs:371:21
    |
371 |                     spi_hdl: spi_handle as *mut _,
    |                     ^^^^^^^ `esp_idf_sys::eth_w5500_config_t` does not have this field
    |
    = note: available fields are: `spi_host_id`, `spi_devcfg`, `int_gpio_num`
```

Installation procedure:

```
install-rust-toolchain.sh -s master
 ./export-esp.sh
cargo build
```

3d-t, 2022-11-21:
OK. I need SPI for SD card. But I have problems installing 4.4 as well.

ivmarkov, 2022-11-21:
> OK. I need SPI for SD card. But I have problems installing 4.4 as well.

Installing the stable version should be much easier than installing the master version. The master version will have all the issues of the stable version plus more. So please do not go in that direction. (I guess I'm repeating myself.)

ivmarkov, 2022-11-21:
Fixed with https://github.com/esp-rs/esp-idf-svc/commit/e20c313f574ee5eacf6ddbc6c2827e961f002d1b

closed 2022-11-21 by ivmarkov