ivmarkov/rust-esp32-std-demo #133: fatal: No names found, cannot describe anything.

3d-t, 2022-11-21:
Try to install demo with ESP-EDF release/v4.4

```
install-rust-toolchain.sh -s release/v4.4
export LIBCLANG_PATH="/Users/maarten/.espressif/tools/xtensa-esp32-elf-clang/esp-15.0.0-20221014-x86_64-apple-darwin/esp-clang/lib/"
export IDF_TOOLS_PATH="/Users/maarten/.espressif"
export IDF_PATH="/Users/maarten/.espressif/frameworks/esp-idf-release-v4.4
. ./export-esp.sh"
cargo build
```

config.toml:
```
build]
# Uncomment the relevant target for your chip here (ESP32, ESP32-S2, ESP32-S3 or ESP32-C3)
target = "xtensa-esp32-espidf"
#target = "xtensa-esp32s2-espidf"
#target = "xtensa-esp32s3-espidf"
#target = "riscv32imc-esp-espidf"

[target.xtensa-esp32-espidf]
linker = "ldproxy"
# Uncomment for ESP-IDF 5 and later (WHEN INSTALLING WITH 4.4 I also need to uncomment)
rustflags = ["--cfg", "espidf_time64"]

[target.xtensa-esp32s2-espidf]
linker = "ldproxy"
# Uncomment for ESP-IDF 5 and later
#rustflags = ["--cfg", "espidf_time64"]

[target.xtensa-esp32s3-espidf]
linker = "ldproxy"
# Uncomment for ESP-IDF 5 and later
#rustflags = ["--cfg", "espidf_time64"]

[target.riscv32imc-esp-espidf]
linker = "ldproxy"
# Uncomment for ESP-IDF 5 and later
#rustflags = ["--cfg", "espidf_time64", "-C", "default-linker-libraries"]
# Future - necessary for the experimental "native build" of esp-idf-sys with ESP32C3
# See also https://github.com/ivmarkov/embuild/issues/16
rustflags = ["-C", "default-linker-libraries"]

[unstable]
build-std = ["std", "panic_abort"]
#build-std-features = ["panic_immediate_abort"] # Only necessary if building against ESP-IDF tag `v4.3.2` (the minimum supported version)

[env]
# Uncomment this and use the esp-idf-sys "native" build feature (`cargo build --features native`) to build against ESP-IDF 5.0 (master)
#ESP_IDF_VERSION = { value = "master" }

# Comment out this when using the PlatformIO build (it only supports `v4.3.2`)
ESP_IDF_VERSION = { value = "release/v4.4" }

# These configurations will pick up your custom "sdkconfig.release", "sdkconfig.debug" or "sdkconfig.defaults[.*]" files
# that you might put in the root of the project
# The easiest way to generate a full "sdkconfig" configuration (as opposed to manually enabling only the necessary flags via "sdkconfig.defaults[.*]"
# is by running "cargo pio espidf menuconfig" (that is, if using the pio builder)
#ESP_IDF_SDKCONFIG = { value = "sdkconfig.release;sdkconfig.debug" }
ESP_IDF_SDKCONFIG_DEFAULTS = { value = "sdkconfig.defaults;sdkconfig.defaults.esp32;sdkconfig.defaults.esp32s2" }
```

Error
```
   Compiling std_detect v0.1.5 (/Users/maarten/.rustup/toolchains/esp/lib/rustlib/src/rust/library/stdarch/crates/std_detect)
   Compiling addr2line v0.16.0
   Compiling std v0.0.0 (/Users/maarten/.rustup/toolchains/esp/lib/rustlib/src/rust/library/std)
The following warnings were emitted during compilation:

warning: Ignoring configuration setting `ESP_IDF_VERSION` (Branch release/v4.4): custom esp-idf repository detected via $IDF_PATH

...

error: failed to run custom build command for `esp-idf-sys v0.31.11`
   Using custom user-supplied esp-idf repository at '/Users/maarten/.espressif/frameworks/esp-idf-release-v4.4' (detected from env variable `IDF_PATH`)
  fatal: No names found, cannot describe anything.
  Using esp-idf v4.4.3 at '/Users/maarten/.espressif/frameworks/esp-idf-release-v4.4'
  fatal: No names found, cannot describe anything.
  thread 'main' panicked at '
  command did not execute successfully, got: exit status: 1
```

ivmarkov, 2022-11-21:
The error is actually `error: failed to run custom build command for esp-idf-sys v0.31.11`

1. To understand where it failed, can you do `export RUST_BACKTRACE=1`, re-run the build, and then paste here the backtrace?
2. Separately from (1) - to workaround the problem - can you try unsetting the `IDF_PATH` environment variable and then re-run the build? The build will then re-download its own ESP IDF copy

ivmarkov, 2022-11-21:
And by the way, you are telling the project that it will be compiled against ESP IDF V5 with this:
```
[target.xtensa-esp32-espidf]
linker = "ldproxy"
# Uncomment for ESP-IDF 5 and later (WHEN INSTALLING WITH 4.4 I also need to uncomment)
rustflags = ["--cfg", "espidf_time64"]
```

Yet, you are trying to compile it (with and without the IDF_PATH variable) against ESP IDF 4.4? Why?

And also: why are you trying to build against ESP IDF 5? It should work, but ESP IDF 5 is not even released yet. I think you should try your first build against the stable ESP IDF version (ESP IDF 4).

ivmarkov, 2022-11-21:
And I see your comment there, but **no**, you definitely do **not** need to uncomment this for ESP IDF 4.4. Quite the opposite.

3d-t, 2022-11-21:
export RUST_BACKTRACE=1
cargo build

```
   Compiling mipidsi v0.5.0
The following warnings were emitted during compilation:

warning: Ignoring configuration setting `ESP_IDF_VERSION` (Branch release/v4.4): custom esp-idf repository detected via $IDF_PATH

error: failed to run custom build command for `esp-idf-sys v0.31.11`

Caused by:
  process didn't exit successfully: `/Users/maarten/dev/rust-esp32-std-demo/target/debug/build/esp-idf-sys-231faf7d37a0ca20/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  cargo:warning=Ignoring configuration setting `ESP_IDF_VERSION` (Branch release/v4.4): custom esp-idf repository detected via $IDF_PATH
  IDF_PYTHON_ENV_PATH=/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env
  PATH=/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin:/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin:/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/tools/cmake/3.23.1/CMake.app/Contents/bin:/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/tools/ninja/1.10.2/:/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env/bin:/Users/maarten/.espressif/frameworks/esp-idf-release-v4.4/tools:$PATH
  Current system platform: macos
  Skipping xtensa-esp32-elf@esp-2021r2-patch5-8.4.0 (already installed)
  Skipping cmake@3.23.1 (already installed)
  Skipping ninja@1.10.2 (already installed)
  Skipping esp32ulp-elf@2.35_20220830 (already installed)
  IDF_PYTHON_ENV_PATH=/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env
  PATH=/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin:/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin:/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/tools/cmake/3.23.1/CMake.app/Contents/bin:/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/tools/ninja/1.10.2/:/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env/bin:/Users/maarten/.espressif/frameworks/esp-idf-release-v4.4/tools:$PATH
  cargo:rerun-if-changed=/Users/maarten/dev/rust-esp32-std-demo/sdkconfig.defaults
  cargo:rerun-if-changed=/Users/maarten/dev/rust-esp32-std-demo/sdkconfig.defaults.esp32
  cargo:rerun-if-changed=/Users/maarten/dev/rust-esp32-std-demo/sdkconfig.defaults.esp32
  cargo:rerun-if-changed=/Users/maarten/dev/rust-esp32-std-demo/sdkconfig.defaults.esp32s2
  CMAKE_PREFIX_PATH_xtensa-esp32-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32-espidf = None
  CMAKE_xtensa_esp32_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: "cmake" "/Users/maarten/dev/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-b253b2b08a4f8a60/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/Users/maarten/.espressif/frameworks/esp-idf-release-v4.4/tools/cmake/toolchain-esp32.cmake" "-DCMAKE_BUILD_TYPE=" "-DPYTHON=/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env/bin/python" "-DCMAKE_INSTALL_PREFIX=/Users/maarten/dev/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-b253b2b08a4f8a60/out" "-DCMAKE_C_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_CXX_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_ASM_FLAGS= -mlongcalls -ffunction-sections -fdata-sections"
  -- Checking Python dependencies...
  Python requirements from /Users/maarten/.espressif/frameworks/esp-idf-release-v4.4/requirements.txt are satisfied.
  -- Project sdkconfig file /Users/maarten/dev/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-b253b2b08a4f8a60/out/sdkconfig
  Loading defaults file /Users/maarten/dev/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-b253b2b08a4f8a60/out/gen-sdkconfig.defaults...
  Loading defaults file /Users/maarten/dev/rust-esp32-std-demo/sdkconfig.defaults...
  Loading defaults file /Users/maarten/dev/rust-esp32-std-demo/sdkconfig.defaults.esp32...
  Loading defaults file /Users/maarten/dev/rust-esp32-std-demo/sdkconfig.defaults.esp32...
  Loading defaults file /Users/maarten/dev/rust-esp32-std-demo/sdkconfig.defaults.esp32s2...
  Loading defaults file /Users/maarten/dev/rust-esp32-std-demo/sdkconfig.defaults.esp32...
  -- App "libespidf" version: 1
  -- Adding linker script /Users/maarten/dev/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-b253b2b08a4f8a60/out/build/esp-idf/esp_system/ld/memory.ld
  -- Adding linker script /Users/maarten/.espressif/frameworks/esp-idf-release-v4.4/components/esp_system/ld/esp32/sections.ld.in
[... cut ...]
     5: build_script_build::native::cargo_driver::build
               at ./build/native/cargo_driver.rs:361:5
     6: build_script_build::native::build
               at ./build/native.rs:17:9
     7: build_script_build::main
               at ./build/build.rs:52:24
     8: core::ops::function::FnOnce::call_once
               at /Users/githubrunner/esp-rs/_work/rust-build/rust-build/library/core/src/ops/function.rs:248:5
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

3d-t, 2022-11-21:
[build]
# Uncomment the relevant target for your chip here (ESP32, ESP32-S2, ESP32-S3 or ESP32-C3)
target = "xtensa-esp32-espidf"
#target = "xtensa-esp32s2-espidf"
#target = "xtensa-esp32s3-espidf"
#target = "riscv32imc-esp-espidf"

[target.xtensa-esp32-espidf]
linker = "ldproxy"
# Uncomment for ESP-IDF 5 and later
#rustflags = ["--cfg", "espidf_time64"]
...

3d-t, 2022-11-21:
As requested:
```
unset IDF_PATH
cargo build
```
Result:
```
   Compiling esp-idf-sys v0.31.11
error: failed to run custom build command for `esp-idf-sys v0.31.11`

Caused by:
  process didn't exit successfully: `/Users/maarten/dev/rust-esp32-std-demo/target/debug/build/esp-idf-sys-231faf7d37a0ca20/build-script-build` (exit status: 101)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  IDF_PYTHON_ENV_PATH=/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env
  PATH=/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin:/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin:/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/tools/cmake/3.23.1/CMake.app/Contents/bin:/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/tools/ninja/1.10.2/:/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env/bin:/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/tools:$PATH
  Current system platform: macos
  Skipping xtensa-esp32-elf@esp-2021r2-patch5-8.4.0 (already installed)
  Skipping cmake@3.23.1 (already installed)
  Skipping ninja@1.10.2 (already installed)
  Skipping esp32ulp-elf@2.35_20220830 (already installed)
  IDF_PYTHON_ENV_PATH=/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env
  PATH=/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin:/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin:/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/tools/cmake/3.23.1/CMake.app/Contents/bin:/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/tools/ninja/1.10.2/:/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env/bin:/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/tools:$PATH
  cargo:rerun-if-changed=/Users/maarten/dev/rust-esp32-std-demo/sdkconfig.defaults
  cargo:rerun-if-changed=/Users/maarten/dev/rust-esp32-std-demo/sdkconfig.defaults.esp32
  cargo:rerun-if-changed=/Users/maarten/dev/rust-esp32-std-demo/sdkconfig.defaults.esp32
  cargo:rerun-if-changed=/Users/maarten/dev/rust-esp32-std-demo/sdkconfig.defaults.esp32s2
  CMAKE_PREFIX_PATH_xtensa-esp32-espidf = None
  CMAKE_PREFIX_PATH_xtensa_esp32_espidf = None
  TARGET_CMAKE_PREFIX_PATH = None
  CMAKE_PREFIX_PATH = None
  CMAKE_xtensa-esp32-espidf = None
  CMAKE_xtensa_esp32_espidf = None
  TARGET_CMAKE = None
  CMAKE = None
  running: "cmake" "/Users/maarten/dev/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-b253b2b08a4f8a60/out" "-G" "Ninja" "-DCMAKE_TOOLCHAIN_FILE=/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/tools/cmake/toolchain-esp32.cmake" "-DCMAKE_BUILD_TYPE=" "-DPYTHON=/Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/python_env/idf4.4_py3.9_env/bin/python" "-DCMAKE_INSTALL_PREFIX=/Users/maarten/dev/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-b253b2b08a4f8a60/out" "-DCMAKE_C_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_CXX_FLAGS= -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections" "-DCMAKE_ASM_FLAGS= -mlongcalls -ffunction-sections -fdata-sections"
  -- Checking Python dependencies...
  Python requirements from /Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/requirements.txt are satisfied.
  -- Project sdkconfig file /Users/maarten/dev/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-b253b2b08a4f8a60/out/sdkconfig
  Loading defaults file /Users/maarten/dev/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-b253b2b08a4f8a60/out/gen-sdkconfig.defaults...
  Loading defaults file /Users/maarten/dev/rust-esp32-std-demo/sdkconfig.defaults...
  Loading defaults file /Users/maarten/dev/rust-esp32-std-demo/sdkconfig.defaults.esp32...
  Loading defaults file /Users/maarten/dev/rust-esp32-std-demo/sdkconfig.defaults.esp32...
  Loading defaults file /Users/maarten/dev/rust-esp32-std-demo/sdkconfig.defaults.esp32s2...
  Loading defaults file /Users/maarten/dev/rust-esp32-std-demo/sdkconfig.defaults.esp32...
  -- App "libespidf" version: 1
  -- Adding linker script /Users/maarten/dev/rust-esp32-std-demo/target/xtensa-esp32-espidf/debug/build/esp-idf-sys-b253b2b08a4f8a60/out/build/esp-idf/esp_system/ld/memory.ld
  -- Adding linker script /Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/components/esp_system/ld/esp32/sections.ld.in
  -- Adding linker script /Users/maarten/dev/rust-esp32-std-demo/.embuild/espressif/esp-idf/release-v4.4/components/esp_rom/esp32/ld/esp32.rom.ld
[... cut ...]
     5: build_script_build::native::cargo_driver::build
               at ./build/native/cargo_driver.rs:361:5
     6: build_script_build::native::build
               at ./build/native.rs:17:9
     7: build_script_build::main
               at ./build/build.rs:52:24
     8: core::ops::function::FnOnce::call_once
               at /Users/githubrunner/esp-rs/_work/rust-build/rust-build/library/core/src/ops/function.rs:248:5
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

3d-t, 2022-11-21:
OK. I removed ~/.rustup and ~/.espressif and opened a new command window just to get default environment vars. Installed all from scratch:
```
git clone https://github.com/esp-rs/rust-build.git
cd rust-build
install-rust-toolchain.sh -s release/v4.4
. ./export-esp.sh
cd .../rust-esp32-std-demo/
cargo clean
rustup default esp
unset CC
unset CXX
export RUST_ESP32_STD_DEMO_WIFI_SSID=[xxx]
export RUST_ESP32_STD_DEMO_WIFI_PASS=[xxx]
cargo build
```
It now compiles:
```
    Finished dev [optimized + debuginfo] target(s) in 0.78s
```
Thanks a lot for your help. I will now try to see if it works with the hardware and I can answer my own request on the installation procedure for MacOS. Maybe some of the issues were caused by switching back and forth between 'master' and 'release/v4.4` or setting /unsetting variables.

3d-t, 2022-11-21:
Same issue with 'target = "xtensa-esp32-espidf"'

3d-t, 2022-11-21:
Flashing seems to work:
```
Chip type:         ESP32 (revision 1)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi, BT, Dual Core, 240MHz, Coding Scheme None
MAC address:       08:3a:f2:7c:38:74
App/part. size:    1528080/4128768 bytes, 37.01%
[00:00:01] ########################################      16/16      segment 0x1000                             [00:00:00] ########################################       1/1       segment 0x8000                             [00:01:32] ########################################     900/900     segment 0x10000                            
Flashing has completed!
```

3d-t, 2022-11-21:
Debug
```
cargo pio espidf monitor /dev/cu.usbserial-0001
```
Output
```
Found compatible PlatformIO Core 6.1.5 -> /Users/maarten/.platformio/penv/bin/platformio
[cargo_pio] platformio.ini not found in /Users/maarten/dev/rust-esp32-std-demo, assuming a Cargo-first project
[embuild::cargo] Found /Users/maarten/dev/rust-esp32-std-demo/.cargo/config.toml
[cargo_pio] Using pre-configured target xtensa-esp32-espidf
[embuild::pio] Configuring MCU 'ESP32' derived from the build target 'xtensa-esp32-espidf'
[embuild::pio] Configuring board 'esp32cam' which supports configured platform 'espressif32', MCU 'ESP32' and configured frameworks [espidf]
[embuild::pio] Resolved platform: 'espressif32', MCU: 'ESP32', board: 'esp32cam', frameworks: [espidf]
Esp32ExceptionDecoder: firmware at /private/var/folders/xl/7zzhxtvx7h338hw63ysm07h00000gn/T/.tmpJYsHHq/proj/.pio/build/debug/firmware.elf does not exist, rebuild the project?
--- Terminal on /dev/cu.usbserial-0001 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
room 4
load��054,len:4800
load:0x40078000,len:17448
```

3d-t, 2022-11-21:
Maybe I need to switch to `v4.3.2`?

config.toml
``` 
# Comment out this when using the PlatformIO build, i.e. `cargo build --features pio` (it only supports `v4.3.2`)
ESP_IDF_VERSION = { value = "release/v4.4" }
```

ivmarkov, 2022-11-21:
No. Please just use `espflash serial-monitor`. In the meantime it works great. (I updated the README so that this is the recommended monitor).

ivmarkov, 2022-11-21:
Are you having **two** boards, one esp32s3 and another - plain esp32?

3d-t, 2022-11-21:
I have a esp32s3 board but I rebuild for plain esp32 and now it seems to work. 

config.toml:
```
[build]
# Uncomment the relevant target for your chip here (ESP32, ESP32-S2, ESP32-S3 or ESP32-C3)
target = "xtensa-esp32-espidf"
#target = "xtensa-esp32s2-espidf"
#target = "xtensa-esp32s3-espidf"
#target = "riscv32imc-esp-espidf"

...

# Comment out this when using the PlatformIO build, i.e. `cargo build --features pio` (it only supports `v4.3.2`)
ESP_IDF_VERSION = { value = "release/v4.4" }
```
Building and flashing:
```
cargo build
espflash /dev/cu.usbserial-0001 target/xtensa-esp32-espidf/debug/rust-esp32-std-demo
espmonitor /dev/cu.usbserial-0001
```
Output:

```
I (680) cpu_start: Pro cpu up.
I (681) cpu_start: Starting app cpu, entry point is 0x40082148
I (0) cpu_start: App cpu up.
I (695) cpu_start: Pro cpu start user code
I (695) cpu_start: cpu freq: 160000000
I (695) cpu_start: Application information:
I (699) cpu_start: Project name:     libespidf
I (704) cpu_start: App version:      1
I (709) cpu_start: Compile time:     Nov 21 2022 14:46:36
I (715) cpu_start: ELF file SHA256:  0000000000000000...
I (721) cpu_start: ESP-IDF:          1f63dc7-dirty
I (727) heap_init: Initializing. RAM available for dynamic allocation:
I (734) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (740) heap_init: At 3FFB7A30 len 000285D0 (161 KiB): DRAM
I (746) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (752) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (759) heap_init: At 400937F8 len 0000C808 (50 KiB): IRAM
I (766) spi_flash: detected chip: generic
I (770) spi_flash: flash io: dio
I (775) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Hello from Rust!
More complex print ["foo", "bar"]
Result: 0, 1
Rust main thread: Thread { id: ThreadId(1), name: Some("main"), .. }
Main TLS before change: 13
Main TLS after change: 42
This is thread number 1, Thread { id: ThreadId(3), name: None, .. }

...

I (158199) rust_esp32_std_demo: A2 sensor reading: 142mV
I (158459) rust_esp32_std_demo: Tick from periodic timer
I (158459) rust_esp32_std_demo: Got message from the event loop: 1669043797.956104s
I (158559) rust_esp32_std_demo: MQTT Message: Received(MessageImpl { id: 0, topic: Some("rust-esp32-std-demo"), details: Complete, data: [78, 111, 119, 32, 105, 115, 32, 49, 54, 54, 57, 48, 52, 51, 55, 57, 55, 46, 57, 53, 54, 49, 48, 52, 115] })
I (159209) rust_esp32_std_demo: Hall sensor reading: 98mV
I (159209) rust_esp32_std_demo: A2 sensor reading: 656mV
I (160219) rust_esp32_std_demo: Hall sensor reading: 99mV
I (160219) rust_esp32_std_demo: A2 sensor reading: 142mV

```

All works now so this issue can be closed. Many thanks for your help.

ivmarkov, 2022-11-21:
Flashing esp32 code for esp32s3 sounds like a very, very bad idea. Don't do that.

ivmarkov, 2022-11-21:
For one, the pin layout of esp32s3 is totally different from the pin layout of esp32. esp32s3 has different peripherals. The instruction set of esp32s3 I think is different from esp32 (but apparently - backward compatible or else the code would not work).

But then, are you sure you have an esp32s3 board? :))
```
Chip type:         ESP32 (revision 1)
```

closed 2022-11-21 by ivmarkov

3d-t, 2022-11-22:
No, it does not have the dual USB so probably ESP.

cwoolum, 2022-11-25:
I'm running into this issue as well. If I comment out `ESP_IDF_VERSION = { value = "release/v4.4" }`, it builds just fine. The problem is that my dependencies required `4.4`. 

Do I need to install a different version of `rustup`?

<details>
  <summary>Build Log with RUST_BACKTRACE enabled</summary>

  ```
error: failed to run custom build command for `esp-idf-sys v0.31.11`

Caused by:
  process didn't exit successfully: `C:\rust-esp32-std-demo\target\debug\build\esp-idf-sys-891ceb2ba450bd37\build-script-build` (exit code: 1)
  --- stdout
  cargo:rerun-if-env-changed=ESP_IDF_TOOLS_INSTALL_DIR
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG
  cargo:rerun-if-env-changed=ESP_IDF_SDKCONFIG_DEFAULTS
  cargo:rerun-if-env-changed=MCU
  cargo:rerun-if-env-changed=ESP_IDF_SYS_ROOT_CRATE
  cargo:rerun-if-env-changed=ESP_IDF_VERSION
  cargo:rerun-if-env-changed=ESP_IDF_REPOSITORY
  cargo:rerun-if-env-changed=ESP_IDF_CMAKE_GENERATOR
  cargo:rerun-if-env-changed=IDF_PATH
  cargo:rerun-if-env-changed=ESP_IDF_COMPONENTS
  IDF_PYTHON_ENV_PATH=C:\rust-esp32-std-demo\.embuild\espressif\python_env\idf4.4_py3.8_env
  PATH=C:\rust-esp32-std-demo\.embuild\espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;C:\rust-esp32-std-demo\.embuild\espressif\tools\cmake\3.23.1\bin;C:\rust-esp32-std-demo\.embuild\espressif\tools\ninja\1.10.2\;C:\rust-esp32-std-demo\.embuild\espressif\python_env\idf4.4_py3.8_env\Scripts;C:\rust-esp32-std-demo\.embuild\espressif\esp-idf\release-v4.4\tools;%PATH%
  Current system platform: win64
  Skipping riscv32-esp-elf@esp-2021r2-patch5-8.4.0 (already installed)
  Skipping cmake@3.23.1 (already installed)
  Skipping ninja@1.10.2 (already installed)
  IDF_PYTHON_ENV_PATH=C:\rust-esp32-std-demo\.embuild\espressif\python_env\idf4.4_py3.8_env
  PATH=C:\rust-esp32-std-demo\.embuild\espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;C:\rust-esp32-std-demo\.embuild\espressif\tools\cmake\3.23.1\bin;C:\rust-esp32-std-demo\.embuild\espressif\tools\ninja\1.10.2\;C:\rust-esp32-std-demo\.embuild\espressif\python_env\idf4.4_py3.8_env\Scripts;C:\rust-esp32-std-demo\.embuild\espressif\esp-idf\release-v4.4\tools;%PATH%

  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: Some(
          [
              "sdkconfig.defaults",
              "sdkconfig.defaults.esp32",
              "sdkconfig.defaults.esp32s2",
          ],
      ),
      mcu: None,
      native: NativeConfig {
          esp_idf_version: Some(
              Branch(
                  "release/v4.4",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: None,
          extra_components: [],
          esp_idf_components: None,
      },
      esp_idf_sys_root_crate: None,
  }
  Using managed esp-idf repository: EspIdfRemote { repo_url: None, git_ref: Branch("release/v4.4") }
  fatal: No names found, cannot describe anything.
  Using esp-idf v4.4.2 at 'C:\rust-esp32-std-demo\.embuild\espressif\esp-idf\release-v4.4'
  fatal: No names found, cannot describe anything.
  error: patch failed: components/app_update/esp_app_desc.c:11
  error: components/app_update/esp_app_desc.c: patch does not apply
  error: patch failed: components/app_update/esp_app_desc.c:11
  error: components/app_update/esp_app_desc.c: patch does not apply
  Error: command '"git" "--git-dir" "C:\\rust-esp32-std-demo\\.embuild\\espressif\\esp-idf\\release-v4.4\\.git" "--work-tree" "C:\\rust-esp32-std-demo\\.embuild\\espressif\\esp-idf\\release-v4.4" "apply" "C:\\Users\\slick\\.cargo\\registry\\src\\github.com-1ecc6299db9ec823\\esp-idf-sys-0.31.11\\patches/esp_app_format_weak_v4.4.diff"' exited with non-zero status code 1

  Stack backtrace:
     0: std::backtrace_rs::backtrace::dbghelp::trace
[... 27 lines cut ...]
    14: std::panicking::try
               at /rustc/70f8737b2f5d3bf7d6b784fad00b663b7ff9feda/library\std\src\panicking.rs:447
    15: std::panic::catch_unwind
               at /rustc/70f8737b2f5d3bf7d6b784fad00b663b7ff9feda/library\std\src\panic.rs:137
    16: std::rt::lang_start_internal::closure$2
               at /rustc/70f8737b2f5d3bf7d6b784fad00b663b7ff9feda/library\std\src\rt.rs:148
    17: std::panicking::try::do_call
               at /rustc/70f8737b2f5d3bf7d6b784fad00b663b7ff9feda/library\std\src\panicking.rs:483
    18: std::panicking::try
               at /rustc/70f8737b2f5d3bf7d6b784fad00b663b7ff9feda/library\std\src\panicking.rs:447
    19: std::panic::catch_unwind
               at /rustc/70f8737b2f5d3bf7d6b784fad00b663b7ff9feda/library\std\src\panic.rs:137
    20: std::rt::lang_start_internal
               at /rustc/70f8737b2f5d3bf7d6b784fad00b663b7ff9feda/library\std\src\rt.rs:148
    21: std::rt::lang_start<enum2$<core::result::Result<tuple$<>,anyhow::Error> > >
               at /rustc/70f8737b2f5d3bf7d6b784fad00b663b7ff9feda\library\std\src\rt.rs:165
    22: main
    23: invoke_main
               at D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:78
    24: __scrt_common_main_seh
               at D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl:288
    25: BaseThreadInitThunk
    26: RtlUserThreadStart
```
</details>