ivmarkov/rust-esp32-std-demo #134: Installation procedure on macOS Big Sur

3d-t, 2022-11-21:
I have been trying to get this demo to work with different ESP_IDF_VERSIONs ('master', 'release/v4.4', 'release/v4.3.2') and different configurations in 'config.toml' but run into various issues as reported in earlier request. Would appreciate to get a step by step instruction on what to do so that the demo runs (for xtensa-esp32-espidf or xtensa-esp32s3-espidf). Something like:

```
rm -rf ~/.rustup/toolchains/esp
rm -rf ~/.espressif/
curl -LO https://github.com/esp-rs/rust-build/releases/download/v1.65.0.1/install-rust-toolchain.sh
chmod a+x install-rust-toolchain.sh
install-rust-toolchain.sh -s release/v4.4
unset CC
unset CXX
export LIBCLANG_PATH="~/.espressif/tools/xtensa-esp32-elf-clang/esp-15.0.0-20221014-x86_64-apple-darwin/esp-clang/lib/"
export IDF_TOOLS_PATH="~/.espressif"
. ./export-esp.sh
export IDF_PATH="~/.espressif/frameworks/esp-idf-release-v4.4"
cargo build
```

ivmarkov, 2022-11-21:
Are you following the build instructions? I specifically am at a loss why you are doing the following:
```
export IDF_TOOLS_PATH="~/.espressif"
```

And then also this:
```
export IDF_PATH="~/.espressif/frameworks/esp-idf-release-v4.4"
```

Also, what `. ./export-esp.sh` does is the same that you do here:
```
export LIBCLANG_PATH="~/.espressif/tools/xtensa-esp32-elf-clang/esp-15.0.0-20221014-x86_64-apple-darwin/esp-clang/lib/"
export IDF_TOOLS_PATH="~/.espressif"
```

so not sure why you are doing it twice, but I think at least this does not hurt.

Now, I'm on Linux, and I take it that the `install-rust-toolchain.sh` script might spit out something different for Mac OS, but if it did *not* spit out a request for you to define (`IDF_TOOLS_PATH` and `IDF_PATH`) *please do not define them*, or else I'm debugging your custom setup rather than vanilla `rust-esp32-std-demo` build. :)

For reference, on Linux `install-rust-toolchain.sh` asks me to put in my ~/.bashrc only these two variables, and this is what I do:
```
export LIBCLANG_PATH=...
export PATH=...
```

3d-t, 2022-11-21:
I opened a new CMD window to eliminate all issues with variables and it now compiles:
```
install-rust-toolchain.sh -s release/v4.4
. ./export-esp.sh
rustup default esp
unset CC
unset CXX
export RUST_ESP32_STD_DEMO_WIFI_SSID=[xxx]
export RUST_ESP32_STD_DEMO_WIFI_PASS=[xxx]
cargo build
```
Thanks again for your help.