ivmarkov/rust-esp32-std-demo #137: Flash and SD card storage examples

3d-t, 2022-11-22:
Can you add or point to an example for:

1. Flash storage (EEPROM, Preferences in .cpp)
2. SD card file reading and writing (SD in the .cpp libraries)?

I have looked at the API's for 'embedded_svc::storage::Storage' and 'edf_sys::sdmmd*' but can not find any examples or test cases.

ivmarkov, 2022-11-22:
Hey, we have a forum (Matrix chat) for these questions, which is noted in the README file of this repo. :)
But I will keep this issue open in case somebody would like to contribute a PR with an example. 
By the way, that might be you - PRs from everyone always appreciated!

3d-t, 2022-11-23:
No problem, will provide if I have something working.