ivmarkov/rust-esp32-std-demo #139: Vanilla build does not run on ESP32-C3

0xa10, 2022-12-18:
I have an ESP32-C3 WROOM on which I am building this project using IDF v4.4 (installed explicitly with espup, since embuild does not seem to want to do a deep clone of the SDK).
The build passes and I manage to flash the results - however Im getting an error regarding "NVS" not being initialized. Here is the build and monitor output:
```
PS C:\Users\alonl\rust-esp32-std-demo> espflash COM7 .\target\riscv32imc-esp-espidf\release\rust-esp32-std-demo -s 4MB --partition-table .\partitions.csv --monitor
New version of espflash is available: v2.0.0-rc.2

Serial port: COM7
Connecting...

Chip type:         ESP32-C3 (revision 3)
Crystal frequency: 40MHz
Flash size:        4MB
Features:          WiFi
MAC address:       58:cf:79:23:a5:a4
App/part. size:    1422848/3145728 bytes, 45.23%
[00:00:00] ########################################      12/12      segment 0x0
[00:00:00] ########################################       1/1       segment 0x8000
[00:00:17] ########################################     809/809     segment 0x10000
Flashing has completed!
Commands:
    CTRL+R    Reset chip
    CTRL+C    Exit

ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4004c97a
0x4004c97a - chip726_phyrom_version_num
    at ??:??
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x172c
load:0x403ce000,len:0x928
0x403ce000 - _iram_text_end
    at ??:??
load:0x403d0000,len:0x2ce0
0x403d0000 - _iram_text_end
    at ??:??
entry 0x403ce000
0x403ce000 - _iram_text_end
    at ??:??
I (24) boot: ESP-IDF v4.4-dev-2825-gb63ec47238 2nd stage bootloader
I (25) boot: compile time 12:10:40
I (25) boot: chip revision: 3
I (28) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (37) boot.esp32c3: SPI Speed      : 80MHz
I (40) boot.esp32c3: SPI Mode       : DIO
I (44) boot.esp32c3: SPI Flash Size : 4MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00300000
I (88) boot: End of partition table
I (92) boot_comm: chip revision: 3, min. application chip revision: 0
I (99) esp_image: segment 0: paddr=00010020 vaddr=3c110020 size=41e58h (269912) map
I (148) esp_image: segment 1: paddr=00051e80 vaddr=3fc8d400 size=02cd4h ( 11476) load
I (150) esp_image: segment 2: paddr=00054b5c vaddr=40380000 size=0b4bch ( 46268) load
I (162) esp_image: segment 3: paddr=00060020 vaddr=42000020 size=1097c8h (1087432) map
I (326) esp_image: segment 4: paddr=001697f0 vaddr=4038b4bc size=01dd0h (  7632) load
I (328) esp_image: segment 5: paddr=0016b5c8 vaddr=50000010 size=00010h (    16) load
I (335) boot: Loaded app from partition at offset 0x10000
I (338) boot: Disabling RNG early entropy source...
I (354) cpu_start: Pro cpu up.
I (362) cpu_start: Pro cpu start user code
I (362) cpu_start: cpu freq: 160000000
I (363) cpu_start: Application information:
I (365) cpu_start: Project name:     libespidf
I (370) cpu_start: App version:      1
I (375) cpu_start: Compile time:     Dec 18 2022 23:27:05
I (381) cpu_start: ELF file SHA256:  0000000000000000...
I (387) cpu_start: ESP-IDF:          v4.4-dirty
I (392) heap_init: Initializing. RAM available for dynamic allocation:
I (399) heap_init: At 3FC94580 len 0002BA80 (174 KiB): DRAM
I (405) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (412) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (419) spi_flash: detected chip: generic
I (423) spi_flash: flash io: dio
I (428) sleep: Configure to isolate all GPIO pins in sleep state
I (434) sleep: Enable automatic switching of GPIO sleep configuration
I (441) cpu_start: Starting scheduler.
Hello from Rust!
More complex print ["foo", "bar"]
Result: 0, 1
Rust main thread: Thread { id: ThreadId(1), name: Some("main"), .. }
Main TLS before change: 13
Main TLS after change: 42
This is thread number 0, Thread { id: ThreadId(2), name: None, .. }
Inner TLS: 0
This is thread number 1, Thread { id: ThreadId(3), name: None, .. }
Inner TLS: 1
This is thread number 2, Thread { id: ThreadId(4), name: None, .. }
Inner TLS: 2
This is thread number 3, Thread { id: ThreadId(5), name: None, .. }
Inner TLS: 3
This is thread number 4, Thread { id: ThreadId(6), name: None, .. }
Inner TLS: 4
About to join the threads. If ESP-IDF was patched successfully, joining will NOT crash
Main TLS after threads: 42
Joins were successful.
I (2570) rust_esp32_std_demo: Testing critical sections
I (2570) rust_esp32_std_demo: Waiting for critical section
I (7570) rust_esp32_std_demo: Critical section acquired
I (7570) pp: pp rom version: 9387209
I (7570) net80211: net80211 rom version: 9387209
I (7580) wifi:wifi driver task: 3fca04b8, prio:23, stack:6656, core=0
I (7580) system_api: Base MAC address is not set
I (7580) system_api: read default base MAC address from EFUSE
I (7590) wifi:wifi firmware version: 7679c42
I (7590) wifi:wifi certification version: v7.0
I (7600) wifi:config NVS flash: disabled
I (7600) wifi:config nano formating: disabled
I (7600) wifi:Init data frame dynamic rx buffer num: 32
I (7610) wifi:Init management frame dynamic rx buffer num: 32
I (7610) wifi:Init management short buffer num: 32
I (7620) wifi:Init dynamic tx buffer num: 32
I (7620) wifi:Init static tx FG buffer num: 2
I (7630) wifi:Init static rx buffer size: 1600
I (7630) wifi:Init static rx buffer num: 10
I (7630) wifi:Init dynamic rx buffer num: 32
I (7640) wifi_init: rx ba win: 6
I (7640) wifi_init: tcpip mbox: 32
I (7650) wifi_init: udp mbox: 6
I (7650) wifi_init: tcp mbox: 6
I (7650) wifi_init: tcp tx win: 5744
I (7660) wifi_init: tcp rx win: 5744
I (7660) wifi_init: tcp mss: 1440
I (7670) wifi_init: WiFi IRAM OP enabled
I (7670) wifi_init: WiFi RX IRAM OP enabled
I (7680) esp_idf_svc::wifi: Driver initialized
I (7680) esp_idf_svc::wifi: Stop requested
I (7690) esp_idf_svc::wifi: Stopping
I (7690) rust_esp32_std_demo: Wifi created, about to scan
I (7700) esp_idf_svc::wifi: About to scan for access points
I (7700) esp_idf_svc::wifi: Disconnect requested
I (7710) esp_idf_svc::wifi: Stop requested
I (7710) esp_idf_svc::wifi: Stopping
I (7720) phy_init: phy_version 907,3369105-dirty,Dec  3 2021,14:55:12
E (7720) phy_init: esp_phy_load_cal_data_from_nvs: NVS has not been initialized. Call nvs_flash_init before starting WiFi/BT.
W (7740) phy_init: failed to load RF calibration data (0x1101), falling back to full calibration
Error:
  × A device attached to the system is not functioning. (os error 31)
```

The device then malfunctions and I have to manually put it into BOOT mode before it presents a working serial interface again.
Any ideas?

ivmarkov, 2022-12-19:
> I have an ESP32-C3 WROOM on which I am building this project using IDF v4.4 (installed explicitly with espup, since embuild does not seem to want to do a deep clone of the SDK). 

Why you need a deep clone in the first place?

> The build passes and I manage to flash the results - however Im getting an error regarding "NVS" not being initialized. Here is the build and monitor output:

Do you have a strong enough power supply to the board?

0xa10, 2022-12-19:
Hi, thanks for the answer
Regarding the deep clone - when building using embuild (instead of having ESP_IDF set by the espup env), the build fails since headers are missing from mbedtls - when looking in `.embuild\espressif\...\v4.4` it seems like the SDK is not being fully cloned. Hence my issue.

Regarding the power supply - Im running it off of a USB port on my PC, which feeds into a standard linear regulator on the board I designed to generate 3.3V.
I'm aware that RF calibration can draw a lot of current but Ive used this design before successfully so I dont think its a power issue.
I've been using this board with MicroPython for the past week or so and its working fine so far...

0xa10, 2022-12-19:
hmm, when using an external USB brick it does seem to power up further since Im now seeing an "aptest" network showing up. However I can't get serial output on the UART pins for some reason, so I'm not sure exactly whats happening - but it probably is power related, thanks.
I'll take a look at the 3.3V rail to see if it breaks down, perhaps adding more bulk capacitance will fix it.

Do you have any idea why this only happens in this setup and not in the MicroPython one? Which I assume also runs RF calibration

0xa10, 2022-12-20:
https://github.com/espressif/esp-idf/issues/8046 - seems to be a pretty accurate description of my issue

closed 2022-12-20 by 0xa10