ivmarkov/rust-esp32-std-demo #144: could not compile esp-idf-svc-0.45.0

xiaoheng14, 2023-02-06:
esp-idf version: 5.0
docker: 
https://hub.docker.com/layers/espressif/idf-rust/esp32c3_latest/images/sha256-b551d30acd5ad97f67df0fef8f3896877639be684667fcab618fc58d09318138?context=explore

esp@ffb182562416:~/rust-esp32-std-demo$ cargo build --release
   Compiling esp-idf-sys v0.32.1
   Compiling esp-idf-hal v0.40.1
   Compiling esp-idf-svc v0.45.0
   Compiling rust-esp32-std-demo v0.29.0 (/home/esp/rust-esp32-std-demo)
error[E0412]: cannot find type `eth_esp32_emac_config_t` in this scope
   --> /home/esp/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.45.0/src/eth.rs:750:62
    |
750 |     fn eth_esp32_emac_default_config(mdc: i32, mdio: i32) -> eth_esp32_emac_config_t {
    |                                                              ^^^^^^^^^^^^^^^^^^^^^^^ help: a struct with a similar name exists: `eth_mac_config_t`
    |
   ::: /home/esp/rust-esp32-std-demo/target/riscv32imc-esp-espidf/release/build/esp-idf-sys-5186fd7313a5cecf/out/bindings.rs:3:1263247
    |
3   | ...(C)] # [derive (Debug , Default , Copy , Clone)] pub struct eth_mac_config_t { # [doc = "< Software reset timeout value (Unit: ms)"] pub sw_reset_timeout_ms : u32 , # [doc = "< Stack size of t...
    |                                                     --------------------------- similarly named struct `eth_mac_config_t` defined here

error[E0422]: cannot find struct, variant or union type `eth_esp32_emac_config_t` in this scope
   --> /home/esp/.cargo/registry/src/github.com-1ecc6299db9ec823/esp-idf-svc-0.45.0/src/eth.rs:751:9
    |
751 |         eth_esp32_emac_config_t {
    |         ^^^^^^^^^^^^^^^^^^^^^^^ help: a struct with a similar name exists: `eth_mac_config_t`
    |
   ::: /home/esp/rust-esp32-std-demo/target/riscv32imc-esp-espidf/release/build/esp-idf-sys-5186fd7313a5cecf/out/bindings.rs:3:1263247
    |
3   | ...(C)] # [derive (Debug , Default , Copy , Clone)] pub struct eth_mac_config_t { # [doc = "< Software reset timeout value (Unit: ms)"] pub sw_reset_timeout_ms : u32 , # [doc = "< Stack size of t...
    |                                                     --------------------------- similarly named struct `eth_mac_config_t` defined here

Some errors have detailed explanations: E0412, E0422.
For more information about an error, try `rustc --explain E0412`.
error: could not compile `esp-idf-svc` due to 2 previous errors

xiaoheng14, 2023-02-08:
comment eth_esp32_emac_config_t makes successful

closed 2023-02-08 by xiaoheng14

katyo, 2023-04-22:
@xiaoheng14 I have same issue