ivmarkov/rust-esp32-std-demo #155: Build issue

yashwanthsinghm, 2023-10-27:
After following steps given in readme getting this issue when im trying to build the project.

```rust-esp32-std-demo> cargo build
error: failed to run `rustc` to learn about target-specific information

Caused by:
  process didn't exit successfully: `rustc - --crate-name ___ --print=file-names --cfg espidf_time64 --target xtensa-esp32-espidf --crate-type bin --crate-type rlib --crate-type dylib --crate-type cdylib --crate-type staticlib --crate-type proc-macro --print=sysroot --print=split-debuginfo --print=crate-name --print=cfg` (exit code: 1)
  --- stderr
  error: Error loading target specification: Could not find specification for target "xtensa-esp32-espidf". Run `rustc --print target-list` for a list of built-in targets```

ivmarkov, 2023-10-27:
You have **not** installed the `espup` toolchain, and you don't have a rustc compiler which can compile for xtensa.
Or, you did it, but you have something which overrides your rust toolchain, so you end up with the stock rustc compiler - be it stable or nightly.

closed 2023-10-27 by ivmarkov

yashwanthsinghm, 2023-10-28:
I have been working on esp nostd hal for a while now so everything is installed up tp date, when I try to build this project I'm getting the issue.

ivmarkov, 2023-10-28:
`rustup default esp`

... or to check what is the current active toolchain:
`rustup show`

Google a bit and try to learn how `rustup` and `rutsc` operate w.r.t. toolchains. Also don't forget before compiling to do `. ~/export-esp.sh` (hopefully that second sourcing step will disappear in the coming months).

I get it that the easiest is to just open a bug here, but you can't imagine what an overhead that is for me.