ivmarkov/rust-esp32-std-demo #159: the `riscv32imc-esp-espidf` target may not be installed

freew01f, 2024-02-22:
I am new in this field,
My PC is Mac m2
rust env info is ....
```
rustup -V
rustup 1.26.0 (5af9b9484 2023-04-05)
info: This is the version for the rustup toolchain manager, not the rustc compiler.
info: The currently active `rustc` version is `rustc 1.78.0-nightly (bb594538f 2024-02-20)`

idf.py --version
ESP-IDF v5.2-37-g93ea06fe3e-dirty

rustup toolchain list
nightly-aarch64-apple-darwin (default)
esp

cargo -V
cargo 1.75.0

rustup component list (ignore not install)
cargo-aarch64-apple-darwin (installed)
rust-src (installed)
rust-std-aarch64-apple-darwin (installed)
rust-std-riscv32imac-unknown-none-elf (installed)
rust-std-riscv32imafc-unknown-none-elf (installed)
rust-std-riscv32imc-unknown-none-elf (installed)
rustc-aarch64-apple-darwin (installed)

rustup target list  (ignore not install)
aarch64-apple-darwin (installed)
riscv32imac-unknown-none-elf (installed)
riscv32imafc-unknown-none-elf (installed)
riscv32imc-unknown-none-elf (installed)

```

when I build this project 
cargo build
I got the the err
```
Compiling peeking_take_while v0.1.2
   Compiling fastrand v2.0.1
   Compiling zero v0.1.3
   Compiling normpath v1.2.0
   Compiling pin-project-lite v0.2.13
   Compiling lazycell v1.3.0
error[E0463]: can't find crate for `core`
  |
  = note: the `riscv32imc-esp-espidf` target may not be installed
  = help: consider downloading the target with `rustup target add riscv32imc-esp-espidf`

error[E0463]: can't find crate for `compiler_builtins`

error[E0463]: can't find crate for `core`
    --> /Users/freewolf/.cargo/registry/src/index.crates.io-6f17d22bba15001f/pin-project-lite-0.2.13/src/lib.rs:1622:9
     |
1622 |     use core::mem::ManuallyDrop;
     |         ^^^^ can't find crate
     |
     = note: the `riscv32imc-esp-espidf` target may not be installed
     = help: consider downloading the target with `rustup target add riscv32imc-esp-espidf`

error[E0463]: can't find crate for `core`
    --> /Users/freewolf/.cargo/registry/src/index.crates.io-6f17d22bba15001f/pin-project-lite-0.2.13/src/lib.rs:1624:13
     |
1624 |     pub use core::{
     |             ^^^^ can't find crate
     |
     = note: the `riscv32imc-esp-espidf` target may not be installed
     = help: consider downloading the target with `rustup target add riscv32imc-esp-espidf`

error[E0405]: cannot find trait `Sized` in this scope
    --> /Users/freewolf/.cargo/registry/src/index.crates.io-6f17d22bba15001f/pin-project-lite-0.2.13/src/lib.rs:1633:32
     |
1633 |     pub struct AlwaysUnpin<T: ?Sized>(PhantomData<T>);
     |                                ^^^^^ not found in this scope

error[E0405]: cannot find trait `Sized` in this scope
    --> /Users/freewolf/.cargo/registry/src/index.crates.io-6f17d22bba15001f/pin-project-lite-0.2.13/src/lib.rs:1635:14
     |
1635 |     impl<T: ?Sized> Unpin for AlwaysUnpin<T> {}
     |              ^^^^^ not found in this scope

error[E0405]: cannot find trait `Sized` in this scope
    --> /Users/freewolf/.cargo/registry/src/index.crates.io-6f17d22bba15001f/pin-project-lite-0.2.13/src/lib.rs:1639:43
     |
1639 |     pub struct UnsafeDropInPlaceGuard<T: ?Sized>(*mut T);
     |                                           ^^^^^ not found in this scope

error[E0405]: cannot find trait `Sized` in this scope
    --> /Users/freewolf/.cargo/registry/src/index.crates.io-6f17d22bba15001f/pin-project-lite-0.2.13/src/lib.rs:1641:14
     |
1641 |     impl<T: ?Sized> UnsafeDropInPlaceGuard<T> {
     |              ^^^^^ not found in this scope

error[E0405]: cannot find trait `Sized` in this scope
    --> /Users/freewolf/.cargo/registry/src/index.crates.io-6f17d22bba15001f/pin-project-lite-0.2.13/src/lib.rs:1648:14
     |
1648 |     impl<T: ?Sized> Drop for UnsafeDropInPlaceGuard<T> {
     |              ^^^^^ not found in this scope

libunwind: malformed __unwind_info at 0x18A5AFCA8 bad second level page
libunwind: malformed __unwind_info at 0x18A5AFCA8 bad second level page
Some errors have detailed explanations: E0405, E0463.
For more information about an error, try `rustc --explain E0405`.
error: could not compile `pin-project-lite` (lib) due to 9 previous errors
warning: build failed, waiting for other jobs to finish...
```

when i install the esp32 target I got
```
rustup target add riscv32imc-esp-espidf
error: toolchain 'nightly-aarch64-apple-darwin' does not contain component 'rust-std' for target 'riscv32imc-esp-espidf'
note: not all platforms have the standard library pre-compiled: https://doc.rust-lang.org/nightly/rustc/platform-support.html
help: consider using `cargo build -Z build-std` instead

cargo build -Z build-std
error: the `-Z` flag is only accepted on the nightly channel of Cargo, but this is the `stable` channel
See https://doc.rust-lang.org/book/appendix-07-nightly-rust.html for more information about Rust release channels.
```

how can I fix this problem, I search alot in google but I can not fix it 
thanks

ivmarkov, 2024-02-22:
You need to use the `esp` Rust toolchain, by fiurst installing it with the `espup` utility, and then swithch it as the default with `rustup default esp`.

OR

you need to use the nightly Rust toolchain with `rustup default nightly`.

OR

you can - in the meantime - also use the stable toolchain, but you need to do a few tricks, so let's leave that for later.

Please follow [this](https://github.com/esp-rs/esp-idf-template).

closed 2024-02-22 by ivmarkov

freew01f, 2024-02-23:
the same

```
rustup show
Default host: aarch64-apple-darwin
rustup home:  /Users/freewolf/.rustup

installed toolchains
--------------------

stable-aarch64-apple-darwin
nightly-aarch64-apple-darwin
esp (default)

active toolchain
----------------

esp (default)
(rustc does not exist)

cargo clean                            
     Removed 2187 files, 746.1MiB total
cargo build                            
   Compiling proc-macro2 v1.0.78
   Compiling unicode-ident v1.0.12
   Compiling libc v0.2.153
   Compiling serde v1.0.197
   Compiling memchr v2.7.1
   Compiling cfg-if v1.0.0
   Compiling crossbeam-utils v0.8.19
   Compiling regex-syntax v0.8.2
   Compiling cc v1.0.86
   Compiling syn v1.0.109
   Compiling glob v0.3.1
   Compiling aho-corasick v1.1.2
   Compiling bitflags v1.3.2
   Compiling rustix v0.38.31
   Compiling autocfg v1.1.0
   Compiling clang-sys v1.7.0
   Compiling quote v1.0.35
   Compiling syn v2.0.50
   Compiling rustversion v1.0.14
   Compiling regex-automata v0.4.5
   Compiling log v0.4.20
   Compiling bitflags v2.4.2
   Compiling semver v1.0.22
   Compiling errno v0.3.8
   Compiling crossbeam-epoch v0.9.18
   Compiling bstr v1.9.0
   Compiling serde_json v1.0.114
   Compiling minimal-lexical v0.2.1
   Compiling same-file v1.0.6
   Compiling thiserror v1.0.57
   Compiling nom v7.1.3
   Compiling walkdir v2.4.0
   Compiling crossbeam-deque v0.8.5
   Compiling nix v0.26.4
   Compiling libloading v0.8.1
   Compiling globset v0.4.14
   Compiling cvt v0.1.2
   Compiling home v0.5.9
   Compiling lazy_static v1.4.0
   Compiling itoa v1.0.10
   Compiling bindgen v0.63.0
   Compiling ryu v1.0.17
   Compiling anyhow v1.0.80
   Compiling either v1.10.0
   Compiling heck v0.4.1
   Compiling which v4.4.2
   Compiling cexpr v0.6.0
[... 556 lines cut ...]
libunwind: malformed __unwind_info at 0x188E83CA8 bad second level page
libunwind: malformed __unwind_info at 0x188E83CA8 bad second level page
libunwind: malformed __unwind_info at 0x188E83CA8 bad second level page
libunwind: malformed __unwind_info at 0x188E83CA8 bad second level page
libunwind: malformed __unwind_info at 0x188E83CA8 bad second level page
libunwind: malformed __unwind_info at 0x188E83CA8 bad second level page
libunwind: malformed __unwind_info at 0x188E83CA8 bad second level page
libunwind: malformed __unwind_info at 0x188E83CA8 bad second level page
libunwind: malformed __unwind_info at 0x188E83CA8 bad second level page
libunwind: malformed __unwind_info at 0x188E83CA8 bad second level page
libunwind: malformed __unwind_info at 0x188E83CA8 bad second level page
libunwind: malformed __unwind_info at 0x188E83CA8 bad second level page
libunwind: malformed __unwind_info at 0x188E83CA8 bad second level page
libunwind: malformed __unwind_info at 0x188E83CA8 bad second level page
libunwind: malformed __unwind_info at 0x188E83CA8 bad second level page
libunwind: malformed __unwind_info at 0x188E83CA8 bad second level page
libunwind: malformed __unwind_info at 0x188E83CA8 bad second level page
libunwind: malformed __unwind_info at 0x188E83CA8 bad second level page
libunwind: malformed __unwind_info at 0x188E83CA8 bad second level page
libunwind: malformed __unwind_info at 0x188E83CA8 bad second level page
libunwind: malformed __unwind_info at 0x188E83CA8 bad second level page
error: could not compile `byteorder` (lib) due to 60 previous errors
```

I already switch esp