ivmarkov/rust-esp32-std-demo #161: Add Wi-Fi callback example

owenthewizard, 2024-03-14:
I'm trying to use [WifiDriver::set_callbacks](https://docs.esp-rs.org/esp-idf-svc/esp_idf_svc/wifi/struct.WifiDriver.html#method.set_callbacks) but I'm not successful. I think an example would be helpful.

owenthewizard, 2024-03-16:
I'm trying to implement this in idiomatic Rust: https://github.com/dokmic/eth2wlan/blob/master/main/main.c

ivmarkov, 2024-03-16:
Did you check the README? Feel free to open this issue at [esp-idf-svc](https://github.com/esp-rs/esp-idf-svc) instead.

With that said, I can't write an example for every use case, as this is supported on my free time. The `esp-idf-*` crates are a community-driven effort, so ideally, folks achieving something open PRs (including for examples) and thus help in driving the crates forwards.

Even describing your issue in more details could be helpful.

closed 2024-03-16 by ivmarkov